hoantv/VNWheel #11: FfbWheel.h question

antigravitacnik, 2020-04-24:
Shouldn't there be in the first two lines **FfbWheel_h** instead of **WHEEL_h** , since the wheel.h doesn't exist as a call in the entire code as far as I could find.
When compiling I get error code **'Wheel_' does not name a type**. So I decided to see what is going on and when I was backtracking the code it looks like that **_USING_HID** cannot be defined, since it is dependable on the **#if defined (USBCON)**, which never appears to be present. Therefore the part with the class definition never becomes part of the code (to my understanding of how the code is supposed to be run and generated).

hoantv, 2020-04-24:
that is my mistake. I made multiple file to test and they have different name. after some copy/paste, i forgot to change :|. 
choose boardtype is Leo or Pro micro, pro. I guess you chose Uno.

antigravitacnik, 2020-04-24:
I chose Mega, but with micro selected it works fine. Will try it out in the following days. Tnx

closed 2020-04-24 by antigravitacnik

hoantv, 2020-04-24:
> 
> 
> I chose Mega, but with micro selected it works fine. Will try it out in the following days. Tnx

With mega, change library HID  i think it work.