hoantv/VNWheel #13: Mismatch between HID descriptor and the struct

meacer, 2020-05-04:
Hi, I was trying to reuse your descriptor in FfbWheelDescriptor.h and I noticed that the button count doesn't match the struct.

The comment here says REPORT_COUNT(8) but the actual descriptor has 0x10 (16):
https://github.com/hoantv/VNWheel/blob/master/FfbWheelDescriptor.h#L45

As a result, Windows ignores reports that use the struct at https://github.com/hoantv/VNWheel/blob/master/WHID.h#L74

The fix is to change uint8_t buttons to uint8_t buttons[2] in that struct (and change the comment to REPORT_COUNT(16)).

hoantv, 2020-05-04:
> Hi, I was trying to reuse your descriptor in FfbWheelDescriptor.h and I noticed that the button count doesn't match the struct.
>
> The comment here says REPORT_COUNT(8) but the actual descriptor has [...]

Ohh, I miss that. At the beginning I use 8 button for the shifter. After I increased 16 forgot change report. I will fix it.

hoantv, 2020-05-08:
hmm, i rollback last work version. Hüseyin Varol tested. it worked.

closed 2020-05-08 by hoantv