hoantv/VNWheel #16: Increase number of buttons dont work

AlvaroWalker, 2020-06-04:
I tried to increase the number of buttons to 16 and 24, but it doesn't work

hoantv, 2020-06-04:
You have to change the descriptor: report size and some function relate to button like
void Wheel_::buttons(uint8_t b) {
  _wheelReport.buttons = b;
}
void Wheel_::press(uint8_t b) {
  _wheelReport.buttons |= (uint8_t)1 << (b - 1);
}
change struct of wheelreport

hoantv, 2020-06-04:
Yes, you have to change from uint8_t to uint16_t.
Next monday, i will release full firmware for stm32 with GUI configurator. Try this :)!

AlvaroWalker, 2020-06-04:
i changed but dont work, i will try more code changes, anyway.

i will wait the stm32 version, will be much better! thanks

closed 2020-06-04 by AlvaroWalker

hoantv, 2020-06-05:
> i changed but dont work, i will try more code changes, anyway.
> 
> i will wait the stm32 version, will be much better! thanks

If you want to join group test firmware. Add my fb: 

https://m.facebook.com/HoanTV