hoantv/VNWheel #17: STM32

TadyTheFish, 2020-11-30:
Does it work with STM32? I can't compile.. USBPluggalbe.h missing. Is this Leonardo specific?

antigravitacnik, 2020-11-30:
Afaik this version is arduino micro/leonardo specific. But there is a STM32 compatible version, but so far I think you have to write to Hoan with your hardware setup so can create .hex file for your setup.

hoantv, 2020-12-01:
I already release fw for stm32. Check it in my repoes

TadyTheFish, 2020-12-01:
You mean the CarSimulatorFirmware?

hoantv, 2020-12-01:
Yes.

TadyTheFish, 2020-12-01:
Ok thank you. I already have MMOS wheel working. I would like to implement a custom wheel. MMOS doesn't dhare his source. I was hoping that you were sharing since the leonardo code is available. But I understand that alot of gours and knowhow is in this project.

Thank you

hoantv, 2020-12-01:
I will share the project but at right time :)

TadyTheFish, 2020-12-01:
NICE :)

hoantv, 2020-12-03:
After it is complete :)!

hoantv, 2020-12-03:
After it is completed. I will shared.

hoantv, 2020-12-03:
I will shared when it is completed.

closed 2020-12-03 by TadyTheFish