hoantv/VNWheel #21: Add support for more Buttons

sisvpfis, 2022-02-28:
Hi,

thank you for your effort in putting together this project! I managed to extend and compile the code and use it to revive an old and no longer driver supported Thrustmaster FFB Wheel. Yet I am struggling to extend the number off buttons/axis supported by the code. I couldn't find the spot where i could rise/change the Number of buttons/axis. As fas as t know there are special axes e.g  for accelerator in the HID... could you help out?

Thank you and my best wishes!
S.

sisvpfis, 2022-03-08:
I made another Step and found where to change the number of buttons from 8 to 16. The HID Description yet is cryptic...
I've tried to change the USAGE_MAX in the descriptor to 16 (0x10) and the REPORT_COUNT as well and changed the variable type from uint8 to uint16 for all the handling functions and HID Struct... I can see the Buttons now but the whole communication seems to have stopped, no input is being outout... Broken HID communication!?

```
  //================================Input Report======================================//
  0x09, 0x01, // USAGE (Pointer)
  // WheelReport
  0x85, HID_REPORTID_WHEEL, // REPORT_ID (1)
  0xA1, 0x00, // COLLECTION (Physical)
  // Button for Shifter
  0x05, 0x09, // USAGE_PAGE  (Button)
  0x19, 0x01, // USAGE_MINIMUM (Button 1)
  0x29, 0x10, // USAGE_MAXIMUM (Button 16)
  0x15, 0x00, // LOGICAL_MINIMUM (0)
  0x25, 0x01, // LOGICAL_MAXIMUM (1)
  0x75, 0x01, // REPORT_SIZE (1)
  0x95, 0x10, // REPORT_COUNT (1)
  0x81, 0x02, //INPUT (Data,Var,Abs)
```

hoantv, 2022-03-21:
You have to change in joystick report. I havent time to maintain this repo :)